7Ji/amlogic-s9xxx-archlinuxarm #1: 系统通过reboot重启后 RTL8761BTV 蓝牙不能使用

xiayang0521, 2022-11-15:
测试盒子为UNT403A，CPU: S905L3A / S905L3A-b

现在的做法和发现：
1. 修改dts源码，在uart_A部分增加bluetooth内容并编译为dtb：
&uart_A {
        status = "okay";
        pinctrl-0 = <&uart_a_pins>, <&uart_a_cts_rts_pins>;
        pinctrl-names = "default";
        uart-has-rtscts;

        bluetooth {
                compatible = "realtek,rtl8723bs-bt";
                enable-gpios = <&gpio GPIOX_17 GPIO_ACTIVE_HIGH>;
                max-speed = <2000000>;
                clocks = <&wifi32k>;
                clock-names = "lpo";
        };
};
临时优化方法：上述代码中bluetooth部分其实只有
compatible = "realtek,rtl8723bs-bt"; 
这一行有用！

2. 将2m波特率的config文件复制到/lib/firmware/rtl_bt/rtl8761b_config.bin。
3. poweroff、断电、重开，蓝牙正常！
4. reboot后蓝牙无法识别无法使用、gpioset命令也无法使用（因为gpio 82已经被系统/rfkill占用）。
参见https://bbs.hassbian.com/thread-18366-1-1.html
该帖子作者试过reset-gpios，不好使，猜测得修改驱动程序源码重新编译内核了。

我在archlinuxarm和Armbian下，遇到的问题是一样，故过来看看有没有啥解决办法。
相关文件已经放在https://gitee.com/xiayang0521/set_bluetooth

7Ji, 2022-11-15:
如果你是修改`uart_A`引用的节点的话，这个节点最后面对应到的是`meson-g12-common.dtsi`的这个节点里的这里

https://github.com/unifreq/linux-6.0.y/blob/bf6e1d88ecfa52f2ffc1105534039b6d107059a4/arch/arm64/boot/dts/amlogic/meson-g12-common.dtsi#L2309

中间，u200开发板的dts引用了`meson-g12a-gpio.h`:
https://github.com/unifreq/linux-6.0.y/blob/bf6e1d88ecfa52f2ffc1105534039b6d107059a4/arch/arm64/boot/dts/amlogic/meson-g12a-u200.dts#L10

https://github.com/unifreq/linux-6.0.y/blob/main/include/dt-bindings/gpio/meson-g12a-gpio.h
其中，`GPIO_X17`在这里定义：
https://github.com/unifreq/linux-6.0.y/blob/bf6e1d88ecfa52f2ffc1105534039b6d107059a4/include/dt-bindings/gpio/meson-g12a-gpio.h#L110

你上面的帖子里和这里的源码是是从radaxa-zero的dts里搬的：
https://github.com/unifreq/linux-6.0.y/blob/bf6e1d88ecfa52f2ffc1105534039b6d107059a4/arch/arm64/boot/dts/amlogic/meson-g12a-radxa-zero.dts#L381
可以参考radaxa-zero做除了这里的其他修改

7Ji, 2022-11-17:
我的所有repo归档直至今年底23考研结束，issue暂时关闭

closed 2022-11-17 by 7Ji

7Ji, 2022-12-27:
我回来了，如果还有问题可以Reopen这个issue

xiayang0521, 2023-02-20:
1.谢谢 我也回来了 gpio toggle pin，加在boot.scr的什么位置？boot.cmd如下
`
echo "Start AMLOGIC mainline U-boot"
if printenv bootfromsd; then exit; fi;
setenv loadaddr "0x44000000"
setenv l_mmc "0 1 2 3"
usb start
for devtype in "usb mmc" ; do
        if test -e ${devtype} ${devnum} uEnv.txt; then
                load ${devtype} ${devnum} ${loadaddr} uEnv.txt
                env import -t ${loadaddr} ${filesize}
                setenv bootargs ${APPEND}
                if printenv mac; then
                        setenv bootargs ${bootargs} mac=${mac}
                elif printenv eth_mac; then
                        setenv bootargs ${bootargs} mac=${eth_mac}
                elif printenv ethaddr; then
                        setenv bootargs ${bootargs} mac=${ethaddr}
                fi
                if load ${devtype} ${devnum} ${kernel_addr_r} ${LINUX}; then
                        if load ${devtype} ${devnum} ${ramdisk_addr_r} ${INITRD}; then
                                if load ${devtype} ${devnum} ${fdt_addr_r} ${FDT}; then
                                        fdt addr ${fdt_addr_r}
                                        booti ${kernel_addr_r} ${ramdisk_addr_r} ${fdt_addr_r}
                                fi
                        fi
                fi
        fi
done
`
2.另外 pin的值是咋算的 我不太清楚，下面我列一下 dts内容，

`/dts-v1/;

/ {
	interrupt-parent = <0x01>;
	#address-cells = <0x02>;
	#size-cells = <0x02>;
	compatible = "cm311,m401a,e900v22c\0amlogic,g12a";
	model = "CM311-1a-YST";

	aliases {
		mmc0 = "/soc/sd@ffe03000";
		mmc1 = "/soc/sd@ffe05000";
		mmc2 = "/soc/mmc@ffe07000";
		serial0 = "/soc/bus@ff800000/serial@3000";
		ethernet0 = "/soc/ethernet@ff3f0000";
	};

	chosen {
		#address-cells = <0x02>;
		#size-cells = <0x02>;
		ranges;
		stdout-path = "serial0:115200n8";

		framebuffer-cvbs {
			compatible = "amlogic,simple-framebuffer\0simple-framebuffer";
			amlogic,pipeline = "vpu-cvbs";
			clocks = <0x02 0xa8 0x02 0x35 0x02 0x3a>;
			status = "disabled";
			power-domains = <0x03 0x00>;
			phandle = <0x59>;
		};

		framebuffer-hdmi {
			compatible = "amlogic,simple-framebuffer\0simple-framebuffer";
			amlogic,pipeline = "vpu-hdmi";
			clocks = <0x02 0xa8 0x02 0x35 0x02 0x3a>;
			status = "disabled";
			power-domains = <0x03 0x00>;
[... 3704 lines cut ...]
		cpu0 = "/cpus/cpu@0";
		cpu1 = "/cpus/cpu@1";
		cpu2 = "/cpus/cpu@2";
		cpu3 = "/cpus/cpu@3";
		l2 = "/cpus/l2-cache0";
		cpu_opp_table = "/opp-table";
		spdif_dit = "/audio-codec-1";
		cvbs_connector_in = "/cvbs-connector/port/endpoint";
		emmc_pwrseq = "/emmc-pwrseq";
		hdmi_connector_in = "/hdmi-connector/port/endpoint";
		flash_1v8 = "/regulator-flash_1v8";
		main_12v = "/regulator-main_12v";
		usb_pwr_en = "/regulator-usb_pwr_en";
		vcc_1v8 = "/regulator-vcc_1v8";
		vcc_3v3 = "/regulator-vcc_3v3";
		vcc_5v = "/regulator-vcc_5v";
		vddao_1v8 = "/regulator-vddao_1v8";
		vddao_3v3 = "/regulator-vddao_3v3";
		vddcpu = "/regulator-vddcpu";
		vddgpu = "/regulator-vddgpu";
		wifi32k = "/wifi32k";
		sdio_pwrseq = "/sdio-pwrseq";
	};
};
`