7Ji/amlogic-s9xxx-archlinuxarm #4: chroot环境域名解析失败

a520ass, 2023-01-06:
主机系统使用的是ArchLinuxARM-aarch64-Amlogic-20221112.2.img.xz，使用arch-chroot 进入chroot环境，无法ping通域名，好像是主机resolv.conf文件的问题，改成/run/systemd/resolve/resolv.conf的软链接就正常了

7Ji, 2023-01-06:
https://github.com/7Ji/amlogic-s9xxx-archlinuxarm/blob/af009a7f4f70e878ce0f749e0ae150311f1a85fa/inroot.sh#L29

镜像创建的时候会启用`systemd-resolved.service`，这个service如果正常运行就会管理一个`/etc/resolv.conf -> /run/systemd/resolve/resolv.conf` 的软链接，并且会覆盖镜像一开始无内容的`/etc/resolv.conf`。这个service是否运行正常？或者你是否自己修改过`/etc/resolv.conf`？如果这个链接不存在的话，那么主机本身也是无法上网的，因为DNS信息直接就不正确。

下面的日志，主客都是这个镜像
```
[nomad7ji@bpi7ji projects]$ sudo losetup -fP ArchLinuxARM-aarch64-Amlogic-20221112.2.img 
[nomad7ji@bpi7ji projects]$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0          7:0    0     2G  0 loop 
├─loop0p1    259:0    0   255M  0 part 
└─loop0p2    259:1    0   1.8G  0 part 
sda            8:0    0 111.8G  0 disk /home/nomad7ji/projects
mmcblk1      179:0    0  58.2G  0 disk /srv
mmcblk2      179:32   0  14.6G  0 disk 
└─mmcblk2p1  179:33   0  14.6G  0 part /
mmcblk2boot0 179:64   0     4M  1 disk 
mmcblk2boot1 179:96   0     4M  1 disk 
[nomad7ji@bpi7ji projects]$ sudo mount -o noatime /dev/loop0p2 /mnt
[nomad7ji@bpi7ji projects]$ sudo mount -o noatime /dev/loop0p1 /mnt/boot/
[nomad7ji@bpi7ji projects]$ sudo arch-chroot /mnt
[root@bpi7ji /]# ping baidu.com -w 1
PING baidu.com (39.156.66.10) 56(84) bytes of data.
64 bytes from 39.156.66.10 (39.156.66.10): icmp_seq=1 ttl=50 time=17.6 ms

--- baidu.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 17.582/17.582/17.582/0.000 ms
```

a520ass, 2023-01-06:
我是直接用的你的镜像，systemd-resolved.service 服务正常，resolv.conf 没改过，默认为空，主机能ping通

[image]

commit 46c378c8 (2023-01-15): inroot.sh: prepare symlink for systemd-resolved

prepare a symbol link /etc/resolv.conf -> /run/systemd/resolve/
resolv.conf, which should be created by systemd-resolved and
utilized by it automatically, but for whatever reason sd-rd
could fail to create it.

fixes #4
closed 2023-01-15 by 7Ji