7Ji/amlogic-s9xxx-archlinuxarm #5: Wifi problem

SajjadSabzkar, 2023-01-07:
Hi , recently i installed your amlogic-archlinuxarm on my TVBOX 
i have problem to connecting to wifi .
any command not exist to help me to connect to wifi such as iwctl , wpa_supplicant , wifi-menu , nmtui ,etc

7Ji, 2023-01-07:
These commands, and further the packages providing them were not pre-installed because that's the Arch way: Keep It Simple, Stupid. You should install the things you need and configure them up as you like and ONLY you should judge whether you need some packages or not and whether some options should be enabled or disabled.

This project only provides the bare minimum to make an ArchLinuxARM system bootable on an Amlogic device and accessible through LAN via SSH: the base alarm package group, the kernel, an AUR helper to upgrade the kernel (yay), openssh, a partition tool for Amlogic's proprietary eMMC partition table (ampart). And ONLY essential options were made to make the box do DHCP on LAN and SSH is enabled to make it accessible through network. **Everything else should be configured by yourself after booting the box**

Read the Arch Wiki on how to connect your box wirelessly: https://wiki.archlinux.org/title/Network_configuration/Wireless

SajjadSabzkar, 2023-01-09:
thanks for your answer , i installed wpa_supplicant manually and it works

closed 2023-02-26 by 7Ji