7Ji/amlogic-s9xxx-archlinuxarm #7: dmesg频繁出现audit: type=...

xiayang0521, 2023-02-23:
机型unt403A cpu S905L3A 使用meson-g12a-s905l3a-cm311.dtb，在Armbian中使用https://gitee.com/xiayang0521/backup_linux_system里的脚本写入了EMMC，安装上ArchLinuxARM-aarch64-Amlogic-20230216_011012-root.tar.xz之后，成功启动，没发现啥使用问题，但dmesg出现如下信息，不知道有没有影响：

...
[   14.133623] kauditd_printk_skb: 18 callbacks suppressed
[   14.133637] audit: type=1334 audit(1675140692.232:36): prog-id=26 op=LOAD
[   14.140124] audit: type=1334 audit(1675140692.236:37): prog-id=27 op=LOAD
[   14.146833] audit: type=1334 audit(1675140692.244:38): prog-id=28 op=LOAD
[   14.332413] audit: type=1130 audit(1675140692.428:39): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=kernel msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   16.606818] platform sdio-pwrseq: deferred probe pending
[   16.606956] platform wifi32k: deferred probe pending
[   18.284162] audit: type=1100 audit(1675140696.380:40): pid=390 uid=0 auid=4294967295 ses=4294967295 subj=kernel msg='op=PAM:authentication grantors=pam_shells,pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   18.304732] audit: type=1101 audit(1675140696.396:41): pid=390 uid=0 auid=4294967295 ses=4294967295 subj=kernel msg='op=PAM:accounting grantors=pam_access,pam_unix,pam_permit,pam_time acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   18.329760] audit: type=1103 audit(1675140696.404:42): pid=390 uid=0 auid=4294967295 ses=4294967295 subj=kernel msg='op=PAM:setcred grantors=pam_shells,pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   18.354489] audit: type=1006 audit(1675140696.404:43): pid=390 uid=0 subj=kernel old-auid=4294967295 auid=0 tty=(none) old-ses=4294967295 ses=1 res=1
[   18.367733] audit: type=1300 audit(1675140696.404:43): arch=c00000b7 syscall=64 success=yes exit=1 a0=3 a1=7fc66cbad0 a2=1 a3=7f9a03e9c0 items=0 ppid=381 pid=390 auid=0 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=1 comm="sshd" exe="/usr/bin/sshd" subj=kernel key=(null)
[   18.393686] audit: type=1327 audit(1675140696.404:43): proctitle=737368643A20726F6F74205B707269765D
[   19.156101] kauditd_printk_skb: 69 callbacks suppressed
[   19.156114] audit: type=1100 audit(1675140697.252:81): pid=393 uid=0 auid=4294967295 ses=4294967295 subj=kernel msg='op=PAM:authentication grantors=pam_shells,pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   19.182016] audit: type=1101 audit(1675140697.256:82): pid=393 uid=0 auid=4294967295 ses=4294967295 subj=kernel msg='op=PAM:accounting grantors=pam_access,pam_unix,pam_permit,pam_time acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   19.206250] audit: type=1110 audit(1675140697.260:83): pid=404 uid=0 auid=0 ses=1 subj=kernel msg='op=PAM:setcred grantors=pam_shells,pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   19.229783] audit: type=1103 audit(1675140697.264:84): pid=393 uid=0 auid=4294967295 ses=4294967295 subj=kernel msg='op=PAM:setcred grantors=pam_shells,pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   19.254846] audit: type=1006 audit(1675140697.264:85): pid=393 uid=0 subj=kernel old-auid=4294967295 auid=0 tty=(none) old-ses=4294967295 ses=3 res=1
[   19.268121] audit: type=1300 audit(1675140697.264:85): arch=c00000b7 syscall=64 success=yes exit=1 a0=3 a1=7fecfd9220 a2=1 a3=7f9c7b49c0 items=0 ppid=381 pid=393 auid=0 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=3 comm="sshd" exe="/usr/bin/sshd" subj=kernel key=(null)
[   19.294091] audit: type=1327 audit(1675140697.264:85): proctitle=737368643A20726F6F74205B707269765D
[   19.303055] audit: type=1110 audit(1675140697.276:86): pid=403 uid=0 auid=0 ses=1 subj=kernel msg='op=PAM:setcred grantors=pam_shells,pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   19.353702] audit: type=1105 audit(1675140697.452:87): pid=393 uid=0 auid=0 ses=3 subj=kernel msg='op=PAM:session_open grantors=pam_loginuid,pam_keyinit,pam_systemd_home,pam_limits,pam_unix,pam_permit,pam_mail,pam_systemd,pam_env acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'
[   19.585025] audit: type=1110 audit(1675140697.680:88): pid=410 uid=0 auid=0 ses=3 subj=kernel msg='op=PAM:setcred grantors=pam_shells,pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sshd" hostname=192.168.1.18 addr=192.168.1.18 terminal=ssh res=success'

xiayang0521, 2023-02-23:
地址 https://gitee.com/xiayang0521/backup_linux_system

7Ji, 2023-02-23:
这个不是报错，这是Arch里默认启用的内核审查功能：
https://wiki.archlinux.org/title/Audit_framework

这个会把任何敏感访问打印到内核日志，可以用来溯源一些安全问题。

如果你要关掉，在启动参数里追加audit=0，`/boot/extlinux/extlinux.conf`或者是`/boot/uEnv.ini`

7Ji, 2023-02-23:
另外，403A的话，建议用我移植的主线bootloader：https://github.com/7Ji/u-boot/releases/tag/v2023.01-unt403a

bootloader写入mmcblk2boot0和mmcblk2boot1 （先`echo 0 > /sys/block/mmcblk2boot0/force_ro` 这样允许写入），再把mmcblk2前4M清空，盒子就能用主线bootloader且不占用eMMC。再给mmcblk2手动分区，可以最大化空间利用，而不是浪费100多M

xiayang0521, 2023-02-23:
> 另外，403A的话，建议用我移植的主线bootloader：https://github.com/7Ji/u-boot/releases/tag/v2023.01-unt403a
> 
> bootloader写入mmcblk2boot0和mmcblk2boot1 （先`echo 0 > /sys/block/mmcblk2boot0/force_ro` 这样允许写入），再把mmcblk2前4M清空，盒子就能用主线bootloader且不占用eMMC。再给mmcblk2手动分区，可以最大化空间利用，而不是浪费100多M

多谢！我来搞一下！

xiayang0521, 2023-02-24:
> 另外，403A的话，建议用我移植的主线bootloader：https://github.com/7Ji/u-boot/releases/tag/v2023.01-unt403a
> 
> bootloader写入mmcblk2boot0和mmcblk2boot1 （先`echo 0 > /sys/block/mmcblk2boot0/force_ro` 这样允许写入），再把mmcblk2前4M清空，盒子就能用主线bootloader且不占用eMMC。再给mmcblk2手动分区，可以最大化空间利用，而不是浪费100多M

谢谢！ 虽然不知道为啥要把bootloader既写入mmcblk2boot0又写入mmcblk2boot1，还是根据你的意见，参照着通常采用的命令把bootloader分别写进了mmcblk2boot0和mmcblk2boot1，在emmc新建ext4分区(从1M开始到最后)，基本完全利用, 写入系统文件，配置好/boot/extlinux/extlinux.conf，成功启动。

7Ji, 2023-02-24:
> 为啥要把bootloader既写入mmcblk2boot0又写入mmcblk2boot1

Aml的SoC在eMMC上找bootloader的顺序是：用户区域开头（mmcblk2）-> 第一个引导区域（mmcblk2boot0）-> 第二个引导区域（mmcblk2boot1）。擦掉mmcblk2前面的之后，就是后面那两个起作用。两个都写有两个目的：
 - 一方面，必然覆盖掉原厂的bootloader，那么就不会启动原厂的bootloader了，反正写0也是写，写主线bootloader也是写，不如写主线bootloader
 - 另一方面，两个都写可以防止其中一个损坏导致无法启动

closed 2023-02-26 by 7Ji