7Ji/amlogic-s9xxx-archlinuxarm #8: 关于linux-headers

xiayang0521, 2023-02-23:
类似的问题也出现在amlogic-s9xxx-armbian，比如我想安装zfs-linux，结果需要linux-headers，

yay -S zfs-linux
 -> Could not find all required packages:
        linux=6.1.12.arch1-1 (Wanted by: zfs-linux)
        linux-headers=6.1.12.arch1-1 (Wanted by: zfs-linux)

能不能在编译内核时给一个对应的程序包？没有这东西似乎编译一个wifi驱动都困难。

xiayang0521, 2023-02-23:
不好意思，在release里看到了，我去试试。

7Ji, 2023-02-23:
你装错包了，zfs-linux是给x86-64的ArchLinux的linux这个内核包编译好的模块，其他内核不能用。arch的树外预编译模块一般是以 模块名-内核名 的格式命名的，比如x86-64下有linux, linux-zen, linux-hardened, linux-lts 几个官方支持的内核，里面只有linux会有这些预编译模块，至多linux-lts也会有，其他的内核都要用dkms模块（比如，我笔记本就是用的linux-zen内核然后nvidia的驱动就是用的nvidia-dkms)：

https://aur.archlinux.org/packages/zfs-dkms

7Ji, 2023-02-23:
另外headers这个包一直是预装的，linux-aarch64-flippy这个aur包提供五个子包：
- linux-aarch64-flippy
- linux-aarch64-flippy-dtb-allwinner
- linux-aarch64-flippy-dtb-amlogic
- linux-aarch64-flippy-dtb-rockchip
- linux-aarch64-flippy-heasers

里面只有全志和瑞芯微的内核不会安装：
https://github.com/7Ji/amlogic-s9xxx-archlinuxarm/blob/master/aur/linux-aarch64-flippy.blacklist

xiayang0521, 2023-02-23:
> 你装错包了，zfs-linux是给x86-64的ArchLinux的linux这个内核包编译好的模块，其他内核不能用。arch的树外预编译模块一般是以 模块名-内核名 的格式命名的，比如x86-64下有linux, linux-zen, linux-hardened, linux-lts 几个官方支持的内核，里面只有linux会有这些预编译模块，至多linux-lts也会有，其他的内核都要用dkms模块（比如，我笔记本就是用的linux-zen内核然后nvidia的驱动就是用的nvidia-dkms)：
> 
> https://aur.archlinux.org/packages/zfs-dkms

好的！多谢！

7Ji, 2023-02-23:
另外，releases里面的*-pkgs.tar.xz也不会包含没有安装的包的，这个文件里打包的是所有装到当次镜像里面的包：
https://github.com/7Ji/alarm-builder-common/blob/e2cd6c9eff7324f6b2a98ee5391b9a14b7d2ffc3/functions/release/archive/archive_pkgs.sh#L7

xiayang0521, 2023-02-23:
> 另外，releases里面的*-pkgs.tar.xz也不会包含没有安装的包的，这个文件里打包的是所有装到当次镜像里面的包： https://github.com/7Ji/alarm-builder-common/blob/e2cd6c9eff7324f6b2a98ee5391b9a14b7d2ffc3/functions/release/archive/archive_pkgs.sh#L7

注意到了 刚才我装headers时 确实覆盖安装。。。

closed 2023-02-23 by 7Ji