7Ji/amlogic-s9xxx-archlinuxarm #9: 新开一个反馈一下unt403a主线u-boot

xiayang0521, 2023-02-24:
发现一个很奇怪的事情使用[unt403a-u-boot.bin.sd.bin](https://github.com/7Ji/u-boot/releases/download/v2023.01-unt403a/unt403a-u-boot.bin.sd.bin)，这个2021年的机子用了可以启动Armbian/alarm
[image]

下面这个2022年的用了启动不了
[image]

还有一个丢了标签的中兴B860AV3.2M，居然也可以用unt403a-u-boot.bin.sd.bin。
另外，你既然有了unt403a，可以参照 https://gitee.com/xiayang0521/set_bluetooth 安装下蓝牙驱动，此外他还有个reboot问题，似乎在这里有人遇到类似问题解决了
https://github.com/ophub/amlogic-s9xxx-armbian/issues/639#issuecomment-1295693361
奈何我没有从源码的dts编译dtb的经验，只做过从dtb反编译改一点再编译回去，所以目前unt403a的蓝牙在reboot之后不能使用的问题还没解决。

7Ji, 2023-02-24:
这堆盒子换皮暗改太多了，有的能用有的不能用是正常现象。我移植的那些标注型号仅限于手里有的。

蓝牙问题，这堆盒子（将近十个）我全部都没用过蓝牙…都是纯粹当提供计算力的服务器用了（distcc的志愿者之类的），暂时也不会去研究怎么修

xiayang0521, 2023-02-24:
不错哦，distcc是嘛，我也搞一个支援一下，有教程链接啥的不？

7Ji, 2023-02-24:
distcc是分布式的c/c++/objective-c编译服务，这个志愿不是说志愿于全球性的活动，而是distcc里面各个参与设备的角色划分：直接执行整个项目编译的是执行者，接收前者分发的任务的是志愿者。

Arch wiki上有文档： https://wiki.archlinux.org/title/Distcc

我主要用来给内核、ungoogled-chromium 之类的大型项目编译提速，一堆垃圾盒子的算力集中起来也可以15分钟多点就能编译完内核