7Ji/amlogic-s9xxx-archlinuxarm #11: unable to up wlan0 interface

SajjadSabzkar, 2023-03-23:
Hi , i after installed your [aur](https://aur.archlinux.org/packages/rtl8189fs-dkms-git) package it successfully installed and wlan0 shown at `ip add` command.
but when i run `sudo ip link set wlan0 up` respond `RTNETLINK Operation not permitted`?

7Ji, 2023-03-23:
What's your kernel version? The stable branch maintainance team at kernel.org backported a troublesome cfg80211 fix which breaks the wlan on all stable branches, for 6.1 this breaks every version after 6.1.15 and until 6.1.18. From 6.1.19 this is fixed but I did not update the kernel package here yet as our upstream (flipply/unifreq) only updated his kernel source to > 6.1.18 yesterday.

7Ji, 2023-03-23:
FYI, I'm still building 6.1.21 for test before pushing that (I test things on my 6 different boxes for each kernel bump), but you can build by your own if you can't wait, apply the following patch to my AUR package [linux-aarch64-flippy](https://aur.archlinux.org/packages/linux-aarch64-flippy) and you should be able to build 6.1.18
```
From e083715ad92d980cd3e4c0f01195547de062cda6 Mon Sep 17 00:00:00 2001
From: 7Ji <pugokushin@gmail.com>
Date: Thu, 23 Mar 2023 14:12:24 +0800
Subject: [PATCH] bump to 6.1.21

---
 .SRCINFO | 14 +++++++-------
 PKGBUILD |  8 ++++----
 config   |  5 +----
 3 files changed, 12 insertions(+), 15 deletions(-)

diff --git a/.SRCINFO b/.SRCINFO
index c227d96..265920a 100644
--- a/.SRCINFO
+++ b/.SRCINFO
@@ -1,5 +1,5 @@
 pkgbase = linux-aarch64-flippy
-       pkgver = 6.1.18
+       pkgver = 6.1.21
        pkgrel = 1
        url = https://github.com/unifreq/linux-6.1.y
        arch = aarch64
@@ -9,11 +9,11 @@ pkgbase = linux-aarch64-flippy
        makedepends = dtc
        makedepends = uboot-tools
        options = !strip
-       source = linux-6.1.y-43a6f6232f8d460d7cb6b5c7d562c94b03300d4f.tar.gz::https://github.com/unifreq/linux-6.1.y/archive/43a6f6232f8d460d7cb6b5c7d562c94b03300d4f.tar.gz
+       source = linux-6.1.y-f9790f631bd69bb7d4d2b21b4547cc224a86f7b7.tar.gz::https://github.com/unifreq/linux-6.1.y/archive/f9790f631bd69bb7d4d2b21b4547cc224a86f7b7.tar.gz
        source = config
        source = linux.preset
-       sha256sums = db7f8bbe3b048a10abd125ece0c4a539e7222030dca1a3c7d6d1ba5e6d9c4da5
-       sha256sums = b64ee3c7ce2a159e19c2b786b40a9031e79a54e330684107c06e89e4e7456077
+       sha256sums = 1dcaf8d500920bf8f3cdde1f5259a85a7fdaec5363689b87e1b43ef968895c58
+       sha256sums = a4c4f35cb6eae42e06985850f294d1278d9332154715dd2df5ff885a25fc05b3
        sha256sums = bdcd6cbf19284b60fac6d6772f1e0ec2e2fe03ce7fe3d7d16844dd6d2b5711f3
 
 pkgname = linux-aarch64-flippy
@@ -36,12 +36,12 @@ pkgname = linux-aarch64-flippy-headers
 
 pkgname = linux-aarch64-flippy-dtb-allwinner
        pkgdesc = DTB files for Allwinner SoCs for flippy's AArch64 kernel
-       provides = linux-aarch64-flippy-dtb=6.1.18
+       provides = linux-aarch64-flippy-dtb=6.1.21
 
 pkgname = linux-aarch64-flippy-dtb-amlogic
        pkgdesc = DTB files for Amlogic SoCs for flippy's AArch64 kernel
-       provides = linux-aarch64-flippy-dtb=6.1.18
+       provides = linux-aarch64-flippy-dtb=6.1.21
 
 pkgname = linux-aarch64-flippy-dtb-rockchip
        pkgdesc = DTB files for Rockchip SoCs for flippy's AArch64 kernel
-       provides = linux-aarch64-flippy-dtb=6.1.18
+       provides = linux-aarch64-flippy-dtb=6.1.21
diff --git a/PKGBUILD b/PKGBUILD
index 3fb352b..2d9145d 100644
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -1,11 +1,11 @@
 # Maintainer: 7Ji <pugokughin@gmail.com>
 
 _desc="flippy's AArch64-focused fork aiming to increase usability"
-_pkgver_main=6.1.18
+_pkgver_main=6.1.21
 _pkgver_suffix=flippy
 _pkgver_uname="${_pkgver_main}-${_pkgver_suffix}"
 _flippy_repo='linux-6.1.y'
-_flippy_commit='43a6f6232f8d460d7cb6b5c7d562c94b03300d4f'
+_flippy_commit='f9790f631bd69bb7d4d2b21b4547cc224a86f7b7'
 _srcname="${_flippy_repo}-${_flippy_commit}"
 
 pkgbase=linux-aarch64-flippy
@@ -31,8 +31,8 @@ source=(
   'linux.preset'
 )
 sha256sums=(
-  'db7f8bbe3b048a10abd125ece0c4a539e7222030dca1a3c7d6d1ba5e6d9c4da5'
-  'b64ee3c7ce2a159e19c2b786b40a9031e79a54e330684107c06e89e4e7456077'
+  '1dcaf8d500920bf8f3cdde1f5259a85a7fdaec5363689b87e1b43ef968895c58'
+  'a4c4f35cb6eae42e06985850f294d1278d9332154715dd2df5ff885a25fc05b3'
   'bdcd6cbf19284b60fac6d6772f1e0ec2e2fe03ce7fe3d7d16844dd6d2b5711f3'
 )
 
diff --git a/config b/config
index 75e5687..025fb12 100644
--- a/config
+++ b/config
@@ -1,6 +1,6 @@
 #
 # Automatically generated file; DO NOT EDIT.
-# Linux/arm64 6.1.18 Kernel Configuration
+# Linux/arm64 6.1.21 Kernel Configuration
 #
 CONFIG_CC_VERSION_TEXT="gcc (GCC) 12.1.0"
 CONFIG_CC_IS_GCC=y
@@ -2865,9 +2865,6 @@ CONFIG_R8125_MULTIPLE_RX_QUEUE=y
 CONFIG_R8125_MULTIPLE_TX_QUEUE=y
 CONFIG_R8125_USE_FIRMWARE_FILE=y
 CONFIG_R8125_PM_SUPPORT=y
-CONFIG_R8125_DOUBLE_VLAN=y
-CONFIG_R8125_PAGE_REUSE=y
-CONFIG_R8125_RX_PACKET_FRAGMENT=y
 CONFIG_R8168=m
 # CONFIG_NET_VENDOR_RENESAS is not set
 # CONFIG_NET_VENDOR_ROCKER is not set
-- 
2.39.2

```

SajjadSabzkar, 2023-03-23:
Of course, I am using Manjaro with kernel 6.2.1.1 and it was able to install the driver that you aurpackaged well. However, apparently the problem must be from dtb, which cannot turn it on.
[log.txt](https://github.com/7Ji/amlogic-s9xxx-archlinuxarm/files/11047685/log.txt)

here some logs from `dmesg` command.

7Ji, 2023-03-23:
```
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 192.168.xxx.xxx/24 metric 1024 brd 192.168.xxx.xxx scope global dynamic wlan0
       valid_lft 891sec preferred_lft 891sec
```

closed 2023-03-23 by 7Ji

7Ji, 2023-03-24:
FYI, I just tagged a new release with the updated kernel
https://github.com/7Ji/amlogic-s9xxx-archlinuxarm/releases/tag/20230323

If you're not bothered to re-install alarm or build the newer kernel, you can just download the `*-pkg.tar.xz` release artifact to get the pre-built kernel pacakge to install manually.

commit 4cbb1e22 (2023-03-23): build/linux-aarch64-flippy: bump to 6.1.21

closes #11