7Ji/amlogic-s9xxx-archlinuxarm #13: 内核包无法编译

Heisenberg5201314, 2023-04-02:
[image]

[tony@alarm linux-aarch64-flippy-bin]$ makepkg -s
==> Making package: linux-aarch64-flippy-bin 6.1.18-1 (Sun 02 Apr 2023 10:43:22 PM CST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Downloading boot-6.1.18-flippy-82+.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
==> ERROR: Failure while downloading https://github.com/ophub/kernel/raw/main/pub/stable/6.1.18/boot-6.1.18-flippy-82+.tar.gz
    Aborting...
[tony@alarm linux-aarch64-flippy-bin]$ 
固件包可以编译 但是内核包似乎地址失效了 无法编译成功

7Ji, 2023-04-03:
-bin的内核包不是这个项目使用的。 而且，包括linux-aarch64-flippy 和 linux-aarch64-flippy-bin 在内，如果有包本身的问题，请到aur包页面去报告。

确实一堆东西都是我一个人维护的，但是你们在相关度低的地方开issue，我追踪问题和标记问题解决变得麻烦更多，后面人有类似问题也不会第一时间在同项目看到解决方案。

对于-bin，这个找不到资源是因为ophub把文件转移到release去了，-bin落后了四个版本我还没有更新，因为我自己用不到。

对于这个项目在用的无后缀非-bin，6.1.22构建是无问题的，如果你不想自己构建，你应该下载本项目的最新release里面的*-pkg.tar.xz再解压，里面是打包当次release里面的*.img.xz和*-root.tar.xz时构建的aur包。pacman -U就可以安装

closed 2023-04-03 by 7Ji

7Ji, 2023-04-03:
https://aur.archlinux.org/cgit/aur.git/commit/?h=linux-aarch64-flippy-bin&id=c79ec7ed5a6129af76a650af7c65f6069b5ac6f6

-bin 的 AUR包也更新到6.1.22了