7Ji/amlogic-s9xxx-archlinuxarm #14: 将启动文件放入EMMC单独分区的办法启动alarm未成功

xiayang0521, 2023-04-10:
我看了https://7ji.github.io/embedded/2023/02/20/gxbb-emmc.html这篇教程，眼前一亮，感觉应该也适用于其他s9xxx的cpu吧，
unt403a未刷主线bootloader，将20230401的alarm烧录U盘启动，分区按
```
ampart /dev/mmcblk2  -m dclone dtb::1M:1 initramfs::32M:1 kernel::64M:1 data::-1:4
===================================================================================
ID| name            |          offset|(   human)|            size|(   human)| masks
-----------------------------------------------------------------------------------
 0: bootloader                      0 (   0.00B)           400000 (   4.00M)      0
    (GAP)                                                 2000000 (  32.00M)
 1: reserved                  2400000 (  36.00M)          4000000 (  64.00M)      0
    (GAP)                                                  800000 (   8.00M)
 2: cache                     6c00000 ( 108.00M)                0 (   0.00B)      0
    (GAP)                                                  800000 (   8.00M)
 3: env                       7400000 ( 116.00M)           800000 (   8.00M)      0
    (GAP)                                                  800000 (   8.00M)
 4: dtb                       8400000 ( 132.00M)           100000 (1024.00K)      1
    (GAP)                                                  800000 (   8.00M)
 5: initramfs                 8d00000 ( 141.00M)          2000000 (  32.00M)      1
    (GAP)                                                  800000 (   8.00M)
 6: kernel                    b500000 ( 181.00M)          4000000 (  64.00M)      1
    (GAP)                                                  800000 (   8.00M)
 7: data                      fd00000 ( 253.00M)        1c4300000 (   7.07G)      4
===================================================================================
```
然后将dtb, initramfs  , kernel  , rootfs写入对应分区，设置了env的config文件，然后执行
```
fw_setenv bootcmd 'run emmc_bootcmd' (教程里是bootcmd_mmc1=run emmc_bootcmd，测试也没成功)
fw_setenv emmc_bootargs 'apt_blkdevs=mmcblk2 zswap.enabled=0 root=/dev/mmcblk2p8 rw'
fw_setenv emmc_dtb_offset '0x42000'
fw_setenv emmc_dtb_count '0x800'
fw_setenv emmc_initramfs_offset '0x46800'
fw_setenv emmc_initramfs_count '0x10000'
fw_setenv emmc_kernel_offset '0x5a800'
fw_setenv emmc_kernel_count '0x20000'
fw_setenv emmc_bootcmd 'mmc dev 1; mmc read ${fdt_addr_r} ${emmc_dtb_offset} ${emmc_dtb_count}; mmc read ${ramdisk_addr_r} ${emmc_initramfs_offset} ${emmc_initramfs_count}; mmc read ${kernel_addr_r} ${emmc_kernel_offset} ${emmc_kernel_count}; setenv bootargs "${emmc_bootargs}"; booti ${kernel_addr_r} ${ramdisk_addr_r} ${fdt_addr_r}'
```
fw_printenv显示变量都已设置成功，关机拔优盘，再次启动后没有成功，不知道哪里出了问题。

7Ji, 2023-04-10:
fdt_addr_r
ramdisk_addr_r
kernel_addr_r
这三个地址是主线u-boot根据内存大小自动生成的，原厂u-boot不会设置这些地址

closed 2023-04-10 by 7Ji

xiayang0521, 2023-04-10:
fw_setenv fdt_addr_r  '0x1000000'
fw_setenv kernel_addr_r  '0x1100000'
fw_setenv ramdisk_addr_r '0x5100000'

7Ji, 2023-04-10:
你有uboot的日志吗

xiayang0521, 2023-04-10:
> 你有uboot的日志吗

主要是没弄工具进uboot。。。

7Ji, 2023-04-10:
而且你为啥要直接裸二进制存储和读取？不能直接一个boot分区一个root分区吗？boot分区格式化为fat32，原厂u-boot就可以读取了

7Ji, 2023-04-10:
dclone一个boot::256M:2 root::-1:4的快照不可以吗

xiayang0521, 2023-04-10:
> 而且你为啥要直接裸二进制存储和读取？不能直接一个boot分区一个root分区吗？boot分区格式化为fat32，原厂u-boot就可以读取了

打算利用一下你的patch里直接读取amlogic分区的功能，如果通过这个可以搞定kernel的启动，后面就直接用amlogic分区了。

dclone一个boot::256M:2 root::-1:4的话，boot分区格式化为fat32，原厂u-boot可以直接读的么？我之前在3.14内核下没成功，我来试试这个哈。

xiayang0521, 2023-04-10:
> 而且你为啥要直接裸二进制存储和读取？不能直接一个boot分区一个root分区吗？boot分区格式化为fat32，原厂u-boot就可以读取了
> 
> 这个博文写的方法是gxbb的s905完全无法用mbr分区表的情况下的无奈之举，哪怕现在我也是用我给主线加的envparts来读写分区，而不是这个方案了。
```
===================================================================================
ID| name            |          offset|(   human)|            size|(   human)| masks
-----------------------------------------------------------------------------------
 0: bootloader                      0 (   0.00B)           400000 (   4.00M)      0
    (GAP)                                                 2000000 (  32.00M)
 1: reserved                  2400000 (  36.00M)          4000000 (  64.00M)      0
    (GAP)                                                  800000 (   8.00M)
 2: cache                     6c00000 ( 108.00M)                0 (   0.00B)      0
    (GAP)                                                  800000 (   8.00M)
 3: env                       7400000 ( 116.00M)           800000 (   8.00M)      0
    (GAP)                                                  800000 (   8.00M)
 4: boot                      8400000 ( 132.00M)         10000000 ( 256.00M)      2
    (GAP)                                                  800000 (   8.00M)
 5: root                     18c00000 ( 396.00M)        1bb400000 (   6.93G)      4
===================================================================================
```
我这里直接进行格式化 mkfs.vfat -n "BOOT_EMMC" /dev/block/boot
这里面有个问题，加载的时候是用fatload mmc 1:5 ... 还是 fatload mmc 1:1 ...？

7Ji, 2023-04-10:
1:5

xiayang0521, 2023-04-10:
> 1:5
也就是说内核加载前，原厂的u-boot可以识别到晶晨的分区哈？

7Ji, 2023-04-10:
> > 1:5
> > 也就是说内核加载前，原厂的u-boot可以识别到晶晨的分区哈？

对

xiayang0521, 2023-04-10:
> 

多谢！我去启动一下！

xiayang0521, 2023-04-11:
> > > 1:5
> > > 也就是说内核加载前，原厂的u-boot可以识别到晶晨的分区哈？
> 
> 对

没试成功啊...回头u-boot下看看，，，，

xiayang0521, 2023-04-13:
> dclone一个`boot::256M:2 root::-1:4`的快照不可以吗

可以了, 启动方式就是一般采用的emmc_autoscript配合uEnv.txt。

但对于镜像里的6.1内核，无论有无加载u-boot.emmc，均无法在emmc启动，（U盘外置系统需要u-boot.ext才能启动，如果删掉，启动不了）。（没用ttl看u-boot输出，真是抹黑乱搞）

我用ophub armbian release里的5.15内核替换掉了6.1内核，则可以采用emmc_autoscript配合uEnv.txt，启动emmc中alarm。