7Ji/amlogic-s9xxx-archlinuxarm #17: MDZ-16-AA启动后网络连接的问题

lochuan, 2023-05-10:
MDZ-16-AA只有一个USB接口，没有有线网络接口
尝试过用HUB同时接U盘和键盘，启动进入系统后再配置无线网络，但使用HUB无法从U盘boot，所以HUB方式也尝试失败
再次尝试ophub/amlogic-s9xxx-armbian，因为这个可以配置first_run_script来启用Wi-Fi，但这个发行版貌似无法驱动MDZ-16-AA的Wi-Fi，通过HDMI视频看booting时候的输出可以看到网络报错

请问有什么解决办法吗？

7Ji, 2023-05-10:
> 使用HUB无法从U盘boot

你使用的是usb3.0的hub吗？gxbb这代的soc用的是dwc的usb3.0控制器去带usb2.0的口，在u-boot下和usb3.0设备的握手会有问题，其中一个表现就是usb3.0 hub

closed 2023-05-10 by 7Ji

lochuan, 2023-05-10:
用的是usb2.0的hub，booting之后我拔下U盘，插入hub能看到识别出来是USB 2.0 Mass Storage，但这个HUB上是带一个百兆网口+3个USB2.0接口，不知道是不是因为这个网口的原因，我试试纯USB HUB

7Ji, 2023-05-17:
有几个不需要机子上同时挂U盘和网卡的解决方案：
1. 线刷 https://github.com/7Ji/amlogic-s9xxx-archlinuxarm/releases/tag/20230207 里面的包，可以直接eMMC启动，那么可以连着有线网卡的情况下调整无线设置
2. 刷着最新release或者action里的镜像，用你手头的其他设备，挂载U盘root分区以后arch-chroot进去，装上包并做好配置。aarch64设备可以原生进去，x86要借助qemu-user-static

不论哪一个，如果要无线，需要 装上wpa_supplicant并填写好wpa_supplicant的设置以及systemd-networkd的设置，允许无线对应的网卡连接