7Ji/amlogic-s9xxx-archlinuxarm #18: WIFI not detected

SajjadSabzkar, 2023-06-03:
Hi , my wifi cheap is rtl88x2cs and my wifi cheap not listed at `ip add` command

7Ji, 2023-06-03:
`rtl88x2cs` is not supported by the mainline kernel and needs a seperate driver.

You could write a PKGBUILD to build with https://github.com/jethome-ru/rtl88x2cs as source, using my [rtl8189fs-dkms-git](https://aur.archlinux.org/packages/rtl8189fs-dkms-git) AUR package as reference.

SajjadSabzkar, 2023-06-03:
this driver already added to the [unifreq Kernel repository](https://github.com/unifreq/linux-6.1.y/tree/main/drivers/net/wireless/realtek/rtl88x2cs) , is right?

7Ji, 2023-06-03:
Yes, it's in unifreq's tree, but I did not enable it, I'll enable that later

SajjadSabzkar, 2023-06-03:
> Yes, it's in unifreq's tree, but I did not enable it, I'll enable that later

can i enable it by myself? what should i  do now ?

7Ji, 2023-06-03:
In case you can't wait till I finish the tests and push the changes to public, you can apply the following patch to the AUR package [linux-aarch64-flippy](https://aur.archlinux.org/packages/linux-aarch64-flippy) and build by yourself:

```
enable-88x2-wireless-driv.patch
From 8685dc5ac6e00c08fd0c2571c9054255e4fc006a Mon Sep 17 00:00:00 2001
From: 7Ji <pugokushin@gmail.com>
Date: Sat, 3 Jun 2023 17:39:27 +0800
Subject: [PATCH] minor bump to 6.1.30/f2a0f, enable 88x2 wireless driver

---
 .SRCINFO | 8 ++++----
 PKGBUILD | 8 ++++----
 config   | 4 ++--
 3 files changed, 10 insertions(+), 10 deletions(-)

diff --git a/.SRCINFO b/.SRCINFO
index 73ac8a4..94b3602 100644
--- a/.SRCINFO
+++ b/.SRCINFO
@@ -1,6 +1,6 @@
 pkgbase = linux-aarch64-flippy
        pkgver = 6.1.30
-       pkgrel = 1
+       pkgrel = 2
        url = https://github.com/unifreq/linux-6.1.y
        arch = aarch64
        license = GPL2
@@ -9,12 +9,12 @@ pkgbase = linux-aarch64-flippy
        makedepends = dtc
        makedepends = uboot-tools
        options = !strip
-       source = linux-6.1.y-7d1218c8616b0c53bcbe099ee219782d2bd579af.tar.gz::https://github.com/unifreq/linux-6.1.y/archive/7d1218c8616b0c53bcbe099ee219782d2bd579af.tar.gz
+       source = linux-6.1.y-f2a0fc7edf04bab82e292f5ca640d3b778dbc6e2.tar.gz::https://github.com/unifreq/linux-6.1.y/archive/f2a0fc7edf04bab82e292f5ca640d3b778dbc6e2.tar.gz
        source = config
        source = linux.preset
        source = 0001-drivers-auxdisplay-openvfd-improve-dev-write-logic.patch
-       sha256sums = 202bc5a74d03320cfa7d53d564f6c6f38ccb8bc31d4fa80168a26c84493d59f6
-       sha256sums = 508b7387380483d7cbe3f541cc1d83cee5f4e333cc9d682868b1f2a5c080257c
+       sha256sums = 1ec11c21e4b9aff733aa11993fa2fe6bd596e45952669a6233c7ab474febd667
+       sha256sums = cca06bf839127aa70d203f939f6072b0972d5c8cc266753b0bf8d7f815c217a8
        sha256sums = bdcd6cbf19284b60fac6d6772f1e0ec2e2fe03ce7fe3d7d16844dd6d2b5711f3
        sha256sums = d4a47c25f5822fe05a473a9ccd8b749429adb23519a72b9043fe42530f990321

diff --git a/PKGBUILD b/PKGBUILD
index 0d70eeb..6769a5a 100644
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -5,7 +5,7 @@ _pkgver_main=6.1.30
 _pkgver_suffix=flippy
 _pkgver_uname="${_pkgver_main}-${_pkgver_suffix}"
 _flippy_repo='linux-6.1.y'
-_flippy_commit='7d1218c8616b0c53bcbe099ee219782d2bd579af'
+_flippy_commit='f2a0fc7edf04bab82e292f5ca640d3b778dbc6e2'
 _srcname="${_flippy_repo}-${_flippy_commit}"

 pkgbase=linux-aarch64-flippy
@@ -17,7 +17,7 @@ pkgname=(
   "${pkgbase}-dtb-rockchip"
 )
 pkgver="${_pkgver_main}"
-pkgrel=1
+pkgrel=2
 arch=('aarch64')
 url="https://github.com/unifreq/${_flippy_repo}"
 license=('GPL2')
@@ -32,8 +32,8 @@ source=(
   '0001-drivers-auxdisplay-openvfd-improve-dev-write-logic.patch'
 )
 sha256sums=(
-  '202bc5a74d03320cfa7d53d564f6c6f38ccb8bc31d4fa80168a26c84493d59f6'
-  '508b7387380483d7cbe3f541cc1d83cee5f4e333cc9d682868b1f2a5c080257c'
+  '1ec11c21e4b9aff733aa11993fa2fe6bd596e45952669a6233c7ab474febd667'
+  'cca06bf839127aa70d203f939f6072b0972d5c8cc266753b0bf8d7f815c217a8'
   'bdcd6cbf19284b60fac6d6772f1e0ec2e2fe03ce7fe3d7d16844dd6d2b5711f3'
   'd4a47c25f5822fe05a473a9ccd8b749429adb23519a72b9043fe42530f990321'
 )
diff --git a/config b/config
index e508abc..6ab5e24 100644
--- a/config
+++ b/config
@@ -1,6 +1,6 @@
 #
 # Automatically generated file; DO NOT EDIT.
-# Linux/arm64 6.1.30 Kernel Configuration
+# Linux/arm64 6.1.31 Kernel Configuration
 #
 CONFIG_CC_VERSION_TEXT="gcc (GCC) 12.1.0"
 CONFIG_CC_IS_GCC=y
@@ -3260,7 +3260,7 @@ CONFIG_RTL8723_COMMON=m
 CONFIG_RTLBTCOEXIST=m
 CONFIG_RTL8XXXU=m
 # CONFIG_RTL8XXXU_UNTESTED is not set
-# CONFIG_RTL8822CS is not set
+CONFIG_RTL8822CS=m
 CONFIG_RTW88=m
 CONFIG_RTW88_CORE=m
 CONFIG_RTW88_PCI=m
--
2.40.1

```

SajjadSabzkar, 2023-06-03:
> In case you can't wait till I finish the tests and push the changes to public, you can apply the following patch to the AUR package [linux-aarch64-flippy](https://aur.archlinux.org/packages/linux-aarch64-flippy) and build by yourself:
> 
> ```
> From 8685dc5ac6e00c08fd0c2571c9054255e4fc006a Mon Sep 17 00:00:00 2001
> From: 7Ji <pugokushin@gmail.com>
> Date: Sat, 3 Jun 2023 17:39:27 +0800
> Subject: [PATCH] minor bump to 6.1.30/f2a0f, enable 88x2 wireless driver
> 
> ---
>  .SRCINFO | 8 ++++----
>  PKGBUILD | 8 ++++----
>  config   | 4 ++--
>  3 files changed, 10 insertions(+), 10 deletions(-)
> 
> diff --git a/.SRCINFO b/.SRCINFO
> index 73ac8a4..94b3602 100644
> --- a/.SRCINFO
> +++ b/.SRCINFO
> @@ -1,6 +1,6 @@
>  pkgbase = linux-aarch64-flippy
>         pkgver = 6.1.30
> -       pkgrel = 1
> +       pkgrel = 2
>         url = https://github.com/unifreq/linux-6.1.y
>         arch = aarch64
>         license = GPL2
> @@ -9,12 +9,12 @@ pkgbase = linux-aarch64-flippy
>         makedepends = dtc
>         makedepends = uboot-tools
>         options = !strip
> -       source = linux-6.1.y-7d1218c8616b0c53bcbe099ee219782d2bd579af.tar.gz::https://github.com/unifreq/linux-6.1.y/archive/7d1218c8616b0c53bcbe099ee219782d2bd579af.tar.gz
> +       source = linux-6.1.y-f2a0fc7edf04bab82e292f5ca640d3b778dbc6e2.tar.gz::https://github.com/unifreq/linux-6.1.y/archive/f2a0fc7edf04bab82e292f5ca640d3b778dbc6e2.tar.gz
>         source = config
>         source = linux.preset
>         source = 0001-drivers-auxdisplay-openvfd-improve-dev-write-logic.patch
> -       sha256sums = 202bc5a74d03320cfa7d53d564f6c6f38ccb8bc31d4fa80168a26c84493d59f6
> -       sha256sums = 508b7387380483d7cbe3f541cc1d83cee5f4e333cc9d682868b1f2a5c080257c
> +       sha256sums = 1ec11c21e4b9aff733aa11993fa2fe6bd596e45952669a6233c7ab474febd667
> +       sha256sums = cca06bf839127aa70d203f939f6072b0972d5c8cc266753b0bf8d7f815c217a8
>         sha256sums = bdcd6cbf19284b60fac6d6772f1e0ec2e2fe03ce7fe3d7d16844dd6d2b5711f3
>         sha256sums = d4a47c25f5822fe05a473a9ccd8b749429adb23519a72b9043fe42530f990321
> 
> diff --git a/PKGBUILD b/PKGBUILD
> index 0d70eeb..6769a5a 100644
> --- a/PKGBUILD
> +++ b/PKGBUILD
> @@ -5,7 +5,7 @@ _pkgver_main=6.1.30
>  _pkgver_suffix=flippy
>  _pkgver_uname="${_pkgver_main}-${_pkgver_suffix}"
>  _flippy_repo='linux-6.1.y'
> -_flippy_commit='7d1218c8616b0c53bcbe099ee219782d2bd579af'
> +_flippy_commit='f2a0fc7edf04bab82e292f5ca640d3b778dbc6e2'
>  _srcname="${_flippy_repo}-${_flippy_commit}"
> 
>  pkgbase=linux-aarch64-flippy
> @@ -17,7 +17,7 @@ pkgname=(
>    "${pkgbase}-dtb-rockchip"
>  )
>  pkgver="${_pkgver_main}"
> -pkgrel=1
> +pkgrel=2
>  arch=('aarch64')
>  url="https://github.com/unifreq/${_flippy_repo}"
>  license=('GPL2')
> @@ -32,8 +32,8 @@ source=(
>    '0001-drivers-auxdisplay-openvfd-improve-dev-write-logic.patch'
>  )
>  sha256sums=(
> -  '202bc5a74d03320cfa7d53d564f6c6f38ccb8bc31d4fa80168a26c84493d59f6'
> -  '508b7387380483d7cbe3f541cc1d83cee5f4e333cc9d682868b1f2a5c080257c'
> +  '1ec11c21e4b9aff733aa11993fa2fe6bd596e45952669a6233c7ab474febd667'
> +  'cca06bf839127aa70d203f939f6072b0972d5c8cc266753b0bf8d7f815c217a8'
>    'bdcd6cbf19284b60fac6d6772f1e0ec2e2fe03ce7fe3d7d16844dd6d2b5711f3'
>    'd4a47c25f5822fe05a473a9ccd8b749429adb23519a72b9043fe42530f990321'
>  )
> diff --git a/config b/config
> index e508abc..6ab5e24 100644
> --- a/config
> +++ b/config
> @@ -1,6 +1,6 @@
>  #
>  # Automatically generated file; DO NOT EDIT.
> -# Linux/arm64 6.1.30 Kernel Configuration
> +# Linux/arm64 6.1.31 Kernel Configuration
>  #
>  CONFIG_CC_VERSION_TEXT="gcc (GCC) 12.1.0"
>  CONFIG_CC_IS_GCC=y
> @@ -3260,7 +3260,7 @@ CONFIG_RTL8723_COMMON=m
>  CONFIG_RTLBTCOEXIST=m
>  CONFIG_RTL8XXXU=m
>  # CONFIG_RTL8XXXU_UNTESTED is not set
> -# CONFIG_RTL8822CS is not set
> +CONFIG_RTL8822CS=m
>  CONFIG_RTW88=m
>  CONFIG_RTW88_CORE=m
>  CONFIG_RTW88_PCI=m
> --
> 2.40.1
> ```

thanks for your response , But I am a beginner in this field, how can I apply this patch?

7Ji, 2023-06-03:
No need to apply that manually, I've pushed the changes to the kernel package, you can either run `yay` to upgrade all of your local AUR packages including `linux-aarch64-flippy`, or wait for the [github action](https://github.com/7Ji/amlogic-s9xxx-archlinuxarm/actions/runs/5164553718) to finish and download the artifact, unzip it, extract the kernel packages from the `*-pkg.tar`, and then `pacman -U` the kernel pacakges

commit 28c27f88 (2023-06-03): build/linux-aarch64-flippy: bump to 6.1.31-1

closes #18
closed 2023-06-03 by 7Ji