7Ji/amlogic-s9xxx-archlinuxarm #21: how to enable openvfd

SajjadSabzkar, 2024-01-26:
Hi , thanks for your amazing project , but how i can enable openvfd ? need to install extra package? or YAopenvfd installed by default?

SajjadSabzkar, 2024-01-26:
My device is X96Air .
> But that's only the userspace part. The kernel driver part only exists in the flippy kernel  (default), but not 7ji kernel. Make sure you've not switched the kernel.

How i can sure which kernel is installed?
i run ```sudo YAopenvfD 0:temp:0 ``` output is ```openvfd_prepare:98: Failed to open '/dev/openvfd' for I/O operation, errno: 2, error: No such file or directory ```

it's seems openvfd not installed yet, right?

7Ji, 2024-01-26:
> How i can sure which kernel is installed

`uname -a`

`pacman -Qs linux-`

7Ji, 2024-01-26:
> i run sudo YAopenvfD 0:temp:0 output is openvfd_prepare:98: Failed to open '/dev/openvfd' for I/O operation, errno: 2, error: No such file or directory 

You can't just do that. `YAopenvfD` is a daemon that's meant to run in background forever. And the `openvfd` kernel module being loaded is one of the assumptions it expects to be already met. Therefore before it could run the module should be loaded with appropriate arguments. The `YAopenvfD.service` systemd unit does all these for you. Run `systemctl start/stop/restart YAopenvfD` and the module would be loaded/unloaded before/after `YAopenvfD`

7Ji, 2024-01-26:
Strange, I just noticed that `openvfd` could not be properly loaded on both of the kernels (technically, it's loaded, but there's no kmsg spam like when I used it last time) and `/dev/openvfd` is therefore missing on them both. I'm feeling some upstream changes have broken the `openvfd` functionality. But I don't have time to bisect the problem for now. Maybe you could try older images from this repo, before I switched to the nightly release model? The old images should still contain old kernels, which I don't preserve locally.

SajjadSabzkar, 2024-01-27:
> Strange, I just noticed that `openvfd` could not be properly loaded on both of the kernels (technically, it's loaded, but there's no kmsg spam like when I used it last time) and `/dev/openvfd` is [...]

Ok thanks for your response , i'll trying to install old images and find what is problem!

closed 2024-04-23 by 7Ji

7Ji, 2024-06-04:
Sorry I just found I removed `openvfd` nodes from device trees from my kernels, I've updated `YAopenvfD` project description and `yaopenvfd` AUR package installation method.

Now, after YAopenvfD v1.1, ensure openvfd module is available, then ensure /openvfd node exists in your device tree, basically follow https://github.com/7Ji/YAopenvfD?tab=readme-ov-file#kernel-dependency , then start YAopenvfD-updater.service and optionally enable autostart following https://github.com/7Ji/YAopenvfD?tab=readme-ov-file#systemd-integration